UniversalRobots/Universal_Robots_ROS_Driver #733: [FEATURE NAME] catkin_make failed due to error 'AnalogOutputType'

shuanglu2023, 2024-12-03:
### Feature summary

Hi everyone. I tried to install the driver on my ubuntu 20.04 system with ros noetic. 
But I failed with catkin_make with the error as follows: 
/home/xxx/catkin_ws_ur5/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1145:57: error: ‘AnalogOutputType’ in namespace ‘urcl’ does not name a type
 1145 |         req.data.pin, req.data.state, static_cast<urcl::AnalogOutputType>(req.data.domain));
Can anyone help me to solve this issue? Thanks!

urfeex, 2024-12-16:
Hi, it seems that you don't have the latest ur_client_library available. Depending on the way you installed the client library you'll either have to make a `sudo apt update && sudo apt dist-upgrade` or pull the `ur_client_library` package in your source tree.

urfeex, 2025-01-15:
Closing this due to a proposed solution without any update. Feel free to reopen if the problem still persists.

closed 2025-01-15 by urfeex