UniversalRobots/Universal_Robots_ROS_Driver #739: UR10e bringup.launch does not print Robot ready to receive control commands

LeoPits, 2025-02-07:
### Affected ROS Driver version(s)

ROS Noetic ubuntu 20.04

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux without realtime patch

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

universal robot version:  5.11 

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

UR10e bringup.launch does not print Robot ready to receive control commands

# Summary
ROS Driver version: noetic, ubuntu 20.04
Affected Robot Software Version(s): 5.11
Affected Robot Hardware Version(s): ur10e

I just followed this repo's README (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver )
and how to setup e-series (except real time settings. but i followedd rest of instruction and there wasn't problem.). 
And I was able to launch bringup, but
it does not print "Robot ready to receive control commands".

## Issue details
after i launch roslaunch ur_robot_driver ur10e_bringup.launch remains in condition `started controllers` but the README say should print  output `Robot ready to receive control commands.`
i try to launch `rosrun ur_robot_driver test_move` but i receive error  `Robot control is currently inactive`
[image]

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-02-07:
Did you start a program with the external control node on the teach pendant? See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver?tab=readme-ov-file#i-started-everything-but-i-cannot-control-the-robot for details.

LeoPits, 2025-02-07:
Yes, the external control it's installed!

[image]

LeoPits, 2025-02-07:
yes, i start the program 
for example i can read `rostopic echo /robot_ur10e/current_pose `
but if i try to publish rostopic
`rostopic pub -r 100 /robot_ur10e/command geometry_msgs/PoseStamped`
the robot remains stuck, in the controller manager i have this situation

[image]

urfeex, 2025-02-10:
Your screenshot shows a typical situation where the robot program is not started. Before trying to send any commands, let's verify that following the steps below work:

1. Start the driver using `roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=<your-robot's-ip>`. That should successfully connect to the robot. From your previous screenshots it seems like this works.
2. Setup your ROS machine's IP address in the robot's installation for the External Control URCap as explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md). Is your ROS machine's IP address really `192.168.56.1`?
3. Generate a program containing the "External Control" node (as you seem to have done)
4. Start that program on the teach pendant. Does it start correctly? Does it keep running?
5. Now you should see the following output in your driver's shell
    ```
    [ INFO] [1739171594.884835872]: Robot requested program
    [ INFO] [1739171594.884904006]: Sent program to robot
    [ INFO] [1739171594.938975129]: Robot connected to reverse interface. Ready to receive control commands.
    ```
6. Now you should be able to use the robot.

If any of the above steps don't succeed, please don't proceed, as not everything will be working. If any of the steps fail, please post the exact commands that you use with their full output.

urfeex, 2025-03-21:
Closing this as there haven't been any updates since my last suggestion. Feel free to comment / reopen if the issue still persist.

closed 2025-03-21 by urfeex