UniversalRobots/Universal_Robots_ROS_Driver #746: ROS Melodic Driver dies on startup

marcomasa, 2025-04-16:
Hi!

First of all, i know melodic is quite outdated, but i need to make the robot arm work on an existing (older) system which still runs melodic. 

When I install from apt, i get this error when i launch the 'ur_robot_driver_node':

```
/opt/ros/melodic/lib/ur_robot_driver/ur_robot_driver_node: symbol lookup error: /opt/ros/melodic/lib/ur_robot_driver/ur_robot_driver_node: undefined symbol: _ZN4urcl8UrDriver10setPayloadEfRKSt5arrayIdLm3EE
```

There also seems to be no valid combination of ur_msgs and the driver that even builds from source.
(tried using apt msgs and src driver aswell, didnt work either).
When I try to build using everything from source, i get this error:

```
In file included from /home/demo/catkin_ws/devel/.private/ur_msgs/include/ur_msgs/GetRobotSoftwareVersion.h:12:0,                                                                                     from /home/demo/catkin_ws/src/ur_repos/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:58,                                                  from /home/demo/catkin_ws/src/ur_repos/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:                                                   /home/demo/catkin_ws/devel/.private/ur_msgs/include/ur_msgs/GetRobotSoftwareVersionResponse.h:27:13: warning: In the GNU C Library, "major" is defined                                by <sys/sysmacros.h>. For historical compatibility, it is                                                                                                                            currently defined by <sys/types.h> as well, but we plan to                                                                                                                           remove this soon. To use "major", include <sys/sysmacros.h>                                                                                                                          directly. If you did not intend to use a system-defined macro                                                                                                                        "major", you should undefine it after including <sys/types.h>.                                                                                                                           : major(0)       
```

Is there a known workaround that sticks?

Thanks so much for your help!

urfeex, 2025-04-17:
First of all: Yes, melodic is no longer supported.

Second: I just tested running the drive using a ROS melodic docker image installing it from apt (`apt install -y ros-melodic-ur-robot-driver`)

That seemed to startup fine. ABI errors like the one you posted usually originate from a partially outdated system. Please try updating your system.

If you want to build from source, you'll have to build all UR packages from source. The noetic branches should work fine. If you want to use the melodic branches, you'll have to go back with the driver to the version of the latest melodic release.

- https://github.com/ros-industrial/universal_robot/tree/noetic
- https://github.com/ros-industrial/ur_msgs/tree/noetic
- https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git

I'm going to close this issue since melodic is unsupported since two years now.

closed 2025-04-17 by urfeex

marcomasa, 2025-04-24:
Thank you for your quick response! :)

Installing from apt did not work for me unfortunately. 
There is still something missing when I try to launch the node (see issue above),
where the "set Payload" part is missing.

I then followed your suggestion using the noetic branches (NOT noetic-devel).

The UR driver in the latest version (tested today with 2.3) still did not build due to this error:

```
In file included from /home/demo/catkin_ws/src/ur_repos/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:0:
/home/demo/catkin_ws/src/ur_repos/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:67:10: 
fatal error: ur_client_library/ur/robot_receive_timeout.h: Datei oder Verzeichnis nicht gefunden
 #include <ur_client_library/ur/robot_receive_timeout.h>
```

But I think I found a solution that sticks finally:

[universal_robot](https://github.com/ros-industrial/universal_robot/tree/noetic) - Branch 'noetic'
[ur_msgs](https://github.com/ros-industrial/ur_msgs/tree/noetic) - Branch 'noetic'
[UR ROS Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git) - Branch 'v2.0'