UniversalRobots/Universal_Robots_ROS_Driver #749: Unable to connect the driver

toolren, 2025-05-19:
When I execute
roslaunch ur_robot_driver ur5_bringup.launch  robot_ip:=192.168.1.10 
This is the complete log file

... logging to /home/zou/.ros/log/2ebb33b2-34d2-11f0-b86f-536638209415/roslaunch-ubuntu-4155.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:39187/

SUMMARY
PARAMETERS

/controller_stopper/consistent_controllers: ['joint_state_con...
/force_torque_sensor_controller/publish_rate: 125
/force_torque_sensor_controller/type: force_torque_sens...
/forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
/forward_cartesian_traj_controller/type: pass_through_cont...
/forward_joint_traj_controller/joints: ['shoulder_pan_jo...
/forward_joint_traj_controller/type: pass_through_cont...
/hardware_control_loop/loop_hz: 125
/joint_based_cartesian_traj_controller/base: base
/joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
/joint_based_cartesian_traj_controller/tip: tool0
/joint_based_cartesian_traj_controller/type: position_controll...
/joint_group_vel_controller/joints: ['shoulder_pan_jo...
/joint_group_vel_controller/type: velocity_controll...
/joint_state_controller/publish_rate: 125
/joint_state_controller/type: joint_state_contr...
/pos_joint_traj_controller/action_monitor_rate: 20
/pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
/pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
/pos_joint_traj_controller/constraints/goal_time: 0.6
/pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
/pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
/pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
/pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
/pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
/pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
/pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
/pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
/pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
/pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
/pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
/pos_joint_traj_controller/joints: ['shoulder_pan_jo...
/pos_joint_traj_controller/state_publish_rate: 125
/pos_joint_traj_controller/stop_trajectory_duration: 0.5
/pos_joint_traj_controller/type: position_controll...
/pose_based_cartesian_traj_controller/base: base
/pose_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
/pose_based_cartesian_traj_controller/tip: tool0_controller
/pose_based_cartesian_traj_controller/type: pose_controllers/...
/robot_description: <?xml version="1....
/robot_status_controller/handle_name: industrial_robot_...
/robot_status_controller/publish_rate: 10
/robot_status_controller/type: industrial_robot_...
/rosdistro: noetic
/rosversion: 1.17.3
/scaled_pos_joint_traj_controller/action_monitor_rate: 20
/scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
/scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
/scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
/scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
/scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
/scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
/scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
/scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
/scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
/scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
[... 212 lines cut ...]
'tcp_offset'
Either your output recipe contains errors or the urcontrol version does not support them.
Failed to connect to robot. Please check if the robot is booted and connected.
Failed to connect to robot. Please check if the robot is booted and connected.
Failed to initialize RTDE client after 3 attempts
Please note that the minimum software version required is 3.12.0 for CB3 robots and 5.5.1 for e-Series robots. The error above could be related to a non-supported polyscope version. Please update your robot's software accordingly.
[ERROR] [1747675006.125459187]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-2] has died!
process has died [pid 4172, exit code 1, cmd /home/zou/catkin_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/zou/.ros/log/2ebb33b2-34d2-11f0-b86f-536638209415/ur_hardware_interface-2.log].
log file: /home/zou/.ros/log/2ebb33b2-34d2-11f0-b86f-536638209415/ur_hardware_interface-2*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-6] killing on exit
[controller_stopper-5] killing on exit
[robot_state_publisher-1] killing on exit
[ros_control_controller_spawner-3] killing on exit
[ur_hardware_interface-2] killing on exit
[ros_control_stopped_spawner-4] killing on exit
[WARN] [1747675007.003772]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1747675007.004724]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_stopper-5] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
I don't know if this is related to my PolyScope version. My PolyScope version is 3.7.1,I'm using a virtual machine-vmware.

urfeex, 2025-05-20:
As our README says:

> Works for all CB3 (with software version >= 3.14.3) and e-Series (software >= 5.9.4) robots and uses the RTDE interface for communication, whenever possible.

In order to use this driver, you should update your robot.

> I'm using a virtual machine-vmware.

We do not recommend running the driver in a virtual machine due to possible performance issues. The driver requires a real-time-like execution and communication with the robot.

toolren, 2025-05-20:
Thank you for your reply. I have also realized that my version is too low. Currently, I am upgrading the version of polyscope. #/