UniversalRobots/Universal_Robots_ROS_Driver #750: [error] [1747390013.934036390]: could not get fresh data package from robot

NANBEI132, 2025-05-20:
### Affected ROS Driver version(s)

ubuntu20.04

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

3.10

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

I have the content of the same ros workspace used by two computers. But the configuration of the computer is not quite the same. It might be different in the driver or some version or something like that. One computer can operate well, but the other one will have such a problem.

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-05-20:
First of all, Robot SW version 3.10 isn't new enough. Please update to 3.14.3 or later. However, if you really are running 3.10, you are probably not using this driver's latest version, as you would probably run into https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/749 in that case.

Without further information, this is rather hard to debug. However, that error usually means one thing:

The robot has an internal control cycle running with 125 Hz (since you seem to be running a CB3 robot, it would be 500 Hz) for an e-series or PolyScope X robot. It sends data to the driver with that frequency.

The driver has a control loop running on the ROS machine. Ideally, that is running with the same frequency. In each control cycle it reads the values sent by the robot. If that read operation times out the error that you mentioned will be printed.

This could potentially result from a non-reliable network connection between the robot and the ROS pc. Do you use a direct network connection between the two?

NANBEI132, 2025-05-22:
Thank you very much. My robotic arm is a UR5, version 3.10. I previously encountered issues when using Wi-Fi, but the problems were resolved after switching to Ethernet. Currently, there's an issue where floating-point data transmission is unsupported. I suspect this problem might be resolved after upgrading the version, and I plan to give it a try then.

NANBEI132, 2025-06-09:
Hello, I've encountered some issues recently. When my computer is connected to the robotic arm, there are error messages on the teaching pendant. However, this problem doesn't occur when using another computer. I'll send you the photos.

发自我的iPhone