UniversalRobots/Universal_Robots_ROS_Driver #754: hardware_interface.h:54:10: fatal error: ur_msgs/SetAnalogOutput.h: 没有那个文件或目录

liu4552, 2025-06-16:
Errors     << ur_robot_driver:make /home/liu/ur5_ws/logs/ur_robot_driver/build.make.000.log
In file included from /home/liu/ur5_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:0:
/home/liu/ur5_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:54:10: fatal error: ur_msgs/SetAnalogOutput.h: 没有那个文件或目录
 #include <ur_msgs/SetAnalogOutput.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ur_robot_driver_plugin.dir/src/hardware_interface.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/liu/ur5_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:0:
/home/liu/ur5_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:54:10: fatal error: ur_msgs/SetAnalogOutput.h: 没有那个文件或目录
 #include <ur_msgs/SetAnalogOutput.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ur_robot_driver_node.dir/src/hardware_interface.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/ur_robot_driver_plugin.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
make[1]: *** [CMakeFiles/ur_robot_driver_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/liu/ur5_ws/build/ur_robot_driver; catkin build --get-env ur_robot_driver | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

..............................................................................
Failed     << ur_robot_driver:make                         [ Exited with code 2 ]
Failed    <<< ur_robot_driver                              [ 20.9 seconds ]   
Abandoned <<< ur_calibration                               [ Unrelated job failed ]
[build] Summary: 24 of 26 packages succeeded.                                 
[build]   Ignored:   None.                                                    
[build]   Warnings:  1 packages succeeded with warnings.                      
[build]   Abandoned: 1 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                       
[build] Runtime: 1 minute and 27.4 seconds total.

ice-cpp, 2025-06-24:
你好，请问解决了吗，我也遇上这个问题了

urfeex, 2025-06-25:
Please keep this conversation in English language.